xueliu/lora-feed #2: Cannot load the JSON package

cretingame, 2018-05-22:
Hi,

Althougth selected the json lua library in menuconfig. It can't load the JSON library. The logread below:
```
Sun Jan  1 02:14:46 2012 user.notice root: SX1301 reset completed
Sun Jan  1 02:14:46 2012 user.notice root: Generate /etc/global_conf.json
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd: lua: /usr/sbin/gen_lora_global_conf:3: module 'JSON' not found:
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no field package.preload['JSON']
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no file './JSON.lua'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no file '/usr/share/lua/JSON.lua'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no file '/usr/share/lua/JSON/init.lua'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no file '/usr/lib/lua/JSON.lua'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no file '/usr/lib/lua/JSON/init.lua'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no file './JSON.so'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no file '/usr/lib/lua/JSON.so'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        no file '/usr/lib/lua/loadall.so'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd: stack traceback:
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        [C]: in function 'require'
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        /usr/sbin/gen_lora_global_conf:3: in main chunk
Sun Jan  1 02:14:46 2012 daemon.notice procd: /etc/rc.d/S99lora_pkt_fwd:        [C]: ?
```
Checking the folders, I got this:

```
root@OpenWrt:~# ls /usr/lib/lua
iwinfo.so          json.lua           luci               mime               mime.so.1.0.3      nixio.so           socket.lua         ubus.so
json               ltn12.lua          lucihttp.so        mime.lua           nixio              socket             socket.so.3.0-rc1
root@OpenWrt:~# ls /usr/lib/lua/json
rpc.lua
```

cretingame, 2018-05-22:
I finally found the module: http://regex.info/code/JSON.lua

closed 2018-05-22 by cretingame

xueliu, 2018-08-18:
Sorry. I did not notice your issue.

I will update this package to use the lua json library in the OpenWRT and push to the mainline.

cretingame, 2018-08-21:
Hi,

No worries, I just included the libraries in my custom openwrt.

Thank you for all the work done, your packages are very convenient to use!