xueliu/lora-feed #3: building OpenWRT within LoRa package.

nhanlee97, 2019-04-22:
Hello xueliu!
I'm having the problem when builing source OpenWRT within lora package. I add lora-package like your guide on github. But when run under command make -j4. It appear an error: " make[3] -C package/libs/libpcap compile
 make[3] -C feeds/lora/basicstation compile
make -r world: build failed. Please re-run make with -j1 V=s to see what's going on
/home/lpnhan/LoRaSetup/chaos_calmer/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 1"
[image]

xueliu, 2019-04-23:
Hello,
I guess that the problem comes from the basicstation. So cancelling the selection of basicstation will be a quick fix.

nhanlee97, 2019-04-23:
Hi xueliu,
Thank for your support. I'll try to do it.