xueliu/lora-feed #6: building lora-packet-forwarder error

liangyuan64, 2019-05-05:
Hi xueliu

openwrt 18.06， It can be compiled the lora_pkt_fwd package
but when I compiled the lora_pkt_fwd package wiht lede 17.01.4, there were many errors. Could you help me solve the problem? Thanks for your time.

make[4]: Entering directory '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1'
make[5]: Entering directory '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1'
make[6]: Entering directory '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1'
make[6]: Leaving directory '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1'
make[6]: Entering directory '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1'
[  7%] Linking C executable lora_pkt_fwd
/home/openwrt/lede-master/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/lib/gcc/mips-openwrt-linux-musl/5.4.0/../../../../mips-openwrt-linux-musl/bin/ld: copy reloc against protected `__stack_chk_guard' is dangerous
CMakeFiles/lora_pkt_fwd.dir/src/lora_pkt_fwd.c.o: In function `thread_valid':
lora_pkt_fwd.c:(.text+0x84): undefined reference to `wait_ms'
CMakeFiles/lora_pkt_fwd.dir/src/lora_pkt_fwd.c.o: In function `parse_SX1301_configuration':
lora_pkt_fwd.c:(.text+0x2de): undefined reference to `lgw_board_setconf'
lora_pkt_fwd.c:(.text+0x4aa): undefined reference to `lgw_lbt_setconf'
lora_pkt_fwd.c:(.text+0x6b6): undefined reference to `lgw_txgain_setconf'
lora_pkt_fwd.c:(.text+0x8ca): undefined reference to `lgw_rxrf_setconf'
lora_pkt_fwd.c:(.text+0x9d6): undefined reference to `lgw_rxif_setconf'
lora_pkt_fwd.c:(.text+0xb2e): undefined reference to `lgw_rxif_setconf'
lora_pkt_fwd.c:(.text+0xc96): undefined reference to `lgw_rxif_setconf'
CMakeFiles/lora_pkt_fwd.dir/src/lora_pkt_fwd.c.o: In function `thread_up':
lora_pkt_fwd.c:(.text+0x1790): undefined reference to `wait_ms'
lora_pkt_fwd.c:(.text+0x17ba): undefined reference to `lgw_receive'
lora_pkt_fwd.c:(.text+0x187c): undefined reference to `lgw_cnt2utc'
lora_pkt_fwd.c:(.text+0x18fc): undefined reference to `lgw_cnt2gps'
CMakeFiles/lora_pkt_fwd.dir/src/lora_pkt_fwd.c.o: In function `$L761':
lora_pkt_fwd.c:(.text+0x2376): undefined reference to `lgw_gps2cnt'
lora_pkt_fwd.c:(.text+0x2732): undefined reference to `lgw_gps2cnt'
CMakeFiles/lora_pkt_fwd.dir/src/lora_pkt_fwd.c.o: In function `thread_gps':
lora_pkt_fwd.c:(.text+0x2dd6): undefined reference to `lgw_parse_ubx'
lora_pkt_fwd.c:(.text+0x2e08): undefined reference to `lgw_gps_get'
lora_pkt_fwd.c:(.text+0x2e1c): undefined reference to `lgw_get_trigcnt'
lora_pkt_fwd.c:(.text+0x2e46): undefined reference to `lgw_gps_sync'
lora_pkt_fwd.c:(.text+0x2e7c): undefined reference to `lgw_parse_nmea'
lora_pkt_fwd.c:(.text+0x2e96): undefined reference to `lgw_gps_get'
CMakeFiles/lora_pkt_fwd.dir/src/lora_pkt_fwd.c.o: In function `thread_jit':
lora_pkt_fwd.c:(.text+0x2fdc): undefined reference to `wait_ms'
lora_pkt_fwd.c:(.text+0x3072): undefined reference to `lgw_status'
lora_pkt_fwd.c:(.text+0x30ec): undefined reference to `lgw_send'
CMakeFiles/lora_pkt_fwd.dir/src/lora_pkt_fwd.c.o: In function `main':
lora_pkt_fwd.c:(.text.startup+0x14): undefined reference to `lgw_version_info'
lora_pkt_fwd.c:(.text.startup+0x134): undefined reference to `lgw_gps_enable'
lora_pkt_fwd.c:(.text.startup+0x30c): undefined reference to `lgw_start'
lora_pkt_fwd.c:(.text.startup+0x43a): undefined reference to `wait_ms'
lora_pkt_fwd.c:(.text.startup+0x8b6): undefined reference to `lgw_get_trigcnt'
lora_pkt_fwd.c:(.text.startup+0xa3e): undefined reference to `lgw_gps_disable'
lora_pkt_fwd.c:(.text.startup+0xa6c): undefined reference to `lgw_stop'
CMakeFiles/lora_pkt_fwd.dir/src/jitqueue.c.o: In function `jit_enqueue':
jitqueue.c:(.text+0x35e): undefined reference to `lgw_time_on_air'
CMakeFiles/lora_pkt_fwd.dir/src/timersync.c.o: In function `thread_timersync':
timersync.c:(.text+0x8e): undefined reference to `lgw_reg_w'
timersync.c:(.text+0xac): undefined reference to `lgw_get_trigcnt'
timersync.c:(.text+0x13a): undefined reference to `lgw_reg_w'
timersync.c:(.text+0x14c): undefined reference to `wait_ms'
collect2: error: ld returned 1 exit status
lora_pkt_fwd/CMakeFiles/lora_pkt_fwd.dir/build.make:198: recipe for target 'lora_pkt_fwd/lora_pkt_fwd-0' failed
make[6]: *** [lora_pkt_fwd/lora_pkt_fwd-0] Error 1
make[6]: Leaving directory '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1'
CMakeFiles/Makefile2:85: recipe for target 'lora_pkt_fwd/CMakeFiles/lora_pkt_fwd.dir/all' failed
make[5]: *** [lora_pkt_fwd/CMakeFiles/lora_pkt_fwd.dir/all] Error 2
make[5]: Leaving directory '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1'
Makefile:127: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1'
Makefile:63: recipe for target '/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1/.built' failed
make[3]: *** [/home/openwrt/lede-master/build_dir/target-mips_24kc_musl-1.1.15/packet_forwarder-4.0.1/.built] Error 2
make[3]: Leaving directory '/home/openwrt/lede-master/feeds/lora/lora-packet-forwarder'
package/Makefile:127: recipe for target 'package/feeds/lora/lora-packet-forwarder/compile' failed
make[2]: *** [package/feeds/lora/lora-packet-forwarder/compile] Error 2
make[2]: Leaving directory '/home/openwrt/lede-master'
package/Makefile:124: recipe for target '/home/openwrt/lede-master/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/openwrt/lede-master/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede-master'
/home/openwrt/lede-master/include/toplevel.mk:193: recipe for target 'world' failed
make: *** [world] Error 2

xueliu, 2019-05-07:
Hello,
This a strange problem. You see this error because Make can not find libloragw it needs. But when you select lora_pkt_fwd libloragw should be automatically selected. So please have a check that libloragw is really selected.

liangyuan64, 2019-05-17:
I'm sure libloragw has been selected ,Trying to recompile but the problem still exists. I'll close comment first, and if I solve this problem, I'll post a solution

xeeee, 2019-05-30:
Does anybody have an update on this?
I'm having the same problem.

I can compile the libloragw fine, but if I had the pkt_fwd I get this errors.

xueliu, 2019-05-31:
> Does anybody have an update on this?
> I'm having the same problem.
> 
> I can compile the libloragw fine, but if I had the pkt_fwd I get this errors.

I will check it in this week.

zimiao815, 2019-05-31:
me too  , 

2, but  name  lora-gateway-hal    !=    libloragw.
 
`Symbol: PACKAGE_lora-gateway-hal [=y]                                                                                                                          |
  | Type  : tristate                                                                                                                                               |
  | Prompt: lora-gateway-hal................... Driver/HAL library for Semtech SX1301                                                                              |
  |   Location:                                                                                                                                                    |
  | (2) -> Libraries                                                                                                                                               |
  |   Defined at tmp/.config-package.in:31488                                                                                                                      |
  |   Depends on: !PACKAGE_lora-picogw-hal [=n]                                                                                                                    |
  |   Selects: PACKAGE_libssp [=n] && PACKAGE_kmod-spi-dev [=y] && PACKAGE_libpthread [=y] && PACKAGE_libc [=y] && PACKAGE_librt [=n]                              |
  |   Selected by: PACKAGE_lora-gateway-hal-tests [=y] && !PACKAGE_lora-picogw-hal [=n] || PACKAGE_lora-gateway-hal-utils [=y] && !PACKAGE_lora-picogw-hal [=n] || `

xueliu, 2019-06-08:
> me too ,
> 
> 2, but name lora-gateway-hal != libloragw.
> 
you are right. Now the library dependence is handled by CMake. It do not matter when lora-gateway-hal != libloragw.
> `Symbol: PACKAGE_lora-gateway-hal [=y] | | Type : tristate | | Prompt: lora-gateway-hal................... Driver/HAL library for Semtech SX1301 | | Location: | | (2) -> Libraries | | Defined at [...]

xueliu, 2019-06-08:
Hello,

Please update the lora-feed repo to the latest version.
Try to run following command:
``` bash
make package/lora-packet-forwarder/clean
make package/lora-packet-forwarder/compile V=s
```
and have a look the CMake initialization part e.g. my compile log:

-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /home/cc/workspace/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc
-- Check for working C compiler: /home/cc/workspace/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/cc/workspace/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++
-- Check for working CXX compiler: /home/cc/workspace/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Found PkgConfig: /home/cc/workspace/openwrt/staging_dir/host/bin/pkg-config (found version "0.29.2")
-- Checking for one of the modules 'loragw'
**-- pkg_config: libloragw is found
-- libloragw include: /home/cc/workspace/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libloragw
-- libloragw library: /home/cc/workspace/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libloragw.so**
-- Configuring done
-- Generating done

to131, 2019-06-11:
Hi,

I try to install openwrt with the package forwarder, but this error appears.

> Package lora-packet-forwarder is missing dependencies for the following libraries:
> libloragw.so.0
> Makefile:63: recipe for target '/home/teste/openwrt/bin/packages/aarch64_cortex-a53/lora/lora-packet-forwarder_4.0.1-1_aarch64_cortex-a53.ipk' failed
> make[3]: *** [/home/teste/openwrt/bin/packages/aarch64_cortex-a53/lora/lora-packet-forwarder_4.0.1-1_aarch64_cortex-a53.ipk] Error 1
> make[3]: Leaving directory '/home/teste/openwrt/feeds/lora/lora-packet-forwarder'
> time: package/feeds/lora/lora-packet-forwarder/compile#0.38#0.26#0.69
> package/Makefile:107: recipe for target 'package/feeds/lora/lora-packet-forwarder/compile' failed
> make[2]: *** [package/feeds/lora/lora-packet-forwarder/compile] Error 2
> make[2]: Leaving directory '/home/teste/openwrt'
> package/Makefile:103: recipe for target '/home/teste/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
> make[1]: *** [/home/teste/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
> make[1]: Leaving directory '/home/teste/openwrt'
> /home/teste/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
> make: *** [world] Error 2

I installed libloragw before the packet forwarder but the libloragw dependency error. I already have the new version of packet forwarder and I already executed it

> make package/lora-packet-forwarder/clean
> make package/lora-packet-forwarder/compile V=s

and nothing happens.

What will be the problem?

Thank you

xeeee, 2019-06-17:
Also having the same problem.
Here's the log,

`-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/bin/mips-openwrt-linux-musl-gcc
-- Check for working C compiler: /home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/bin/mips-openwrt-linux-musl-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/bin/mips-openwrt-linux-musl-g++
-- Check for working CXX compiler: /home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/bin/mips-openwrt-linux-musl-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Found PkgConfig: /home/user/Documents/openwrt/lorawan/openwrt/staging_dir/host/bin/pkg-config (found version "0.29.2") 
-- Checking for one of the modules 'loragw'
-- pkg_config: libloragw is found
-- libloragw include: /home/user/Documents/openwrt/lorawan/openwrt/staging_dir/target-mips_24kc_musl/usr/include/libloragw
-- libloragw library: 
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_ARG1
    CMAKE_FIND_ROOT_PATH_MODE_LIBRARY
    CMAKE_MODULE_LINKER_FLAGS
    CMAKE_SHARED_LINKER_FLAGS
    DL_LIBRARY

-- Build files have been written to: /home/user/Documents/openwrt/lorawan/openwrt/build_dir/target-mips_24kc_musl/packet_forwarder-4.0.1
touch /home/user/Documents/openwrt/lorawan/openwrt/build_dir/target-mips_24kc_musl/packet_forwarder-4.0.1/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/user/Documents/openwrt/lorawan/openwrt/build_dir/target-mips_24kc_musl/packet_forwarder-4.0.1/.built
touch /home/user/Documents/openwrt/lorawan/openwrt/build_dir/target-mips_24kc_musl/packet_forwarder-4.0.1/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/user/Documents/openwrt/lorawan/openwrt/build_dir/target-mips_24kc_musl/packet_forwarder-4.0.1:packet_forwarder-4.0.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/include -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include/fortify -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/user/Documents/openwrt/lorawan/openwrt/build_dir/target-mips_24kc_musl/packet_forwarder-4.0.1:packet_forwarder-4.0.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/include -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include/fortify -I/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include " LDFLAGS="-L/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/lib -L/home/user/Documents/openwrt/lorawan/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/lib -znow -zrelro " make -j1 -C /home/user/Documents/openwrt/lorawan/openwrt/build_dir/target-mips_24kc_musl/packet_forwarder-4.0.1/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/user/Documents/openwrt/lorawan/openwrt/build_dir/target-mips_24kc_musl/packet_forwarder-4.0.1:packet_forwarder-4.0.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-mus [... cut ...]
time: package/feeds/lora/lora-packet-forwarder/compile#2.29#0.33#2.73
package/Makefile:107: recipe for target 'package/feeds/lora/lora-packet-forwarder/compile' failed
make[1]: *** [package/feeds/lora/lora-packet-forwarder/compile] Error 2
make[1]: Leaving directory '/home/user/Documents/openwrt/lorawan/openwrt'
/home/user/Documents/openwrt/lorawan/openwrt/include/toplevel.mk:216: recipe for target 'package/lora-packet-forwarder/compile' failed
make: *** [package/lora-packet-forwarder/compile] Error 2`

xueliu, 2019-06-21:
> Hi,
>
> I try to install openwrt with the package forwarder, but this error appears.
>
>
> I installed libloragw before the packet forwarder but the libloragw dependency error. I already have the new [...]

Hi, from the error log I guess that you did not use the latest code. You may clean the build tree and try a again.

xueliu, 2019-06-21:
> Also having the same problem.
> Here's the log,
> https://www.dropbox.com/s/rdd4ajdr04vpl89/loragw.txt

It is strange that the line 

> -- libloragw library: 

is empty.

xueliu, 2019-06-21:
I guess that the problem may be dependence conflict of lora-packet-forwarder between the mainline and my feed.
My patch for lora-gateway-hal is already accepted by OpenWRT. Its library package calles libloragw. In order to support the pico gateway in this local feed and avoid the name conflict, I change the library name to the lora-gateway-hal and also the dependence of lora-packet-forwarder. But the offical lora-gateway-hal does not been changed and it has a higher priority. So If you use the latest official feed and together with this lora-feed. The compile of lora-packet-forwarder will complain the missing of lora-gateway-hal (libloragw).

In order to solve the problem, I add patches to revert the lora-gateway-hal library name back to libloragw. I will change the library name of pico-hal to libpicogw to avoid the conflict.

so please clean, update and try again.

xeeee, 2019-06-30:
I tried it again this morning with a clean build and I'm still having the same issue.

pacmenDE, 2019-07-13:
hi, i´ve tried too to compile the packet_forwarder. But i get the same errors descriped in this issue. Anybody knows how to resolve it?
Thanks Patrik

pacmenDE, 2019-07-15:
Here is my solution for the lora gateway and packet forwarder....
https://github.com/pacmenDE/openwrt_MT7688_loragw_pkt_forwarder

xeeee, 2019-07-16:
Hi, how did you managed to fix the issue?
I need to compile it for another architecture.

xueliu, 2019-07-16:
> Hi, how did you managed to fix the issue?
> I need to compile it for another architecture.

Which OpenWRT version do you use ?

xueliu, 2019-07-17:
Hello all,

I finally reproduced this problem in OpenWRT 18.06.04. But I really don't know why CMake pkg_search_module() only found the head files but no libraries. So I add a third method (Findloragw.cmake) to locate the headers and libraries. I hope this time the patch can work.

closed 2019-07-17 by xueliu

pacmenDE, 2019-07-17:
Hi,
i hope, i can help to resolve the problem for compile the packet_forwarder!

i´am not a developer, but i´ve found a functional openwrt V15.05 on github from rakwireless. 
https://github.com/RAKWireless/RAK2247-RAK833-LoRaGateway-OpenWRT-MT7628
with copy and paste solution i am able to produce openwrt 15.05 with difference architecture running lora GW und pkt_fwd.

I tried to compile with V17 and V18 but i get on both versions many compile errors too!

I hope i can help you with my tip and you can resolve your problem.:-)
Patrik

commit 1bf0cd12 (2019-07-17): lora-packet-forwarder: fix #6 for OpenWRT 18.04.x