xueliu/lora-feed #7: /dev/ttyACM0 OpenWRT compatibility/installation

donetdisturb, 2019-05-11:
Hello Xue, thank you for your work.

I own this concentrator card: https://www.n-fuse.co/devices/LoRaWAN-Concentrator-Card-mini-PCIe.html

Following what is written in: https://www.semtech.com/uploads/documents/picocell_gateway_user_guide.pdf
it works perfectly on rPi and Ubuntu, via host driver/HAL USB, like a standard Semtech PicoCell GW.

Instead of using /dev/SPI is it possible to use /dev/ACM0 on OpenWRT?

I tried to install the only dependent kernel module (kmod-usb-acm) and the peripheral is correctly recognized, but the packet forwarder goes wrong.

xueliu, 2019-05-12:
Hello,
As you can see your concentrator card is picocell-gateway. It needs picocell gateway hal https://github.com/Lora-net/picoGW_hal and picocell packet forwarder https://github.com/Lora-net/picoGW_packet_forwarder. But both of them are currently not supported in lora-feed.

I can add support for them in a few days. Since I don't have picocell gateway hardware, I can not test the packet forwarder and its configurations. So your feedback will be very usefull.

donetdisturb, 2019-05-12:
Hi Xiu, sure, I can try and give the necessary feedback.
With this addition there would be many LoRa + WRT implementations! Thank you so much.

xueliu, 2019-05-12:
Hello,
My work is partially finished. Both pico-hal and pico packet forward should be compilable. But you should copy your configuration file to the right position. And there is currently no Luci app for it.

donetdisturb, 2019-05-13:
Hello Xue, good news: it works!
I have compiled a snapshot for Alix x86 with the following packages:
lora-picogw-hal-tests
lora-picogw-packet-forwarder
lora-picogw-packet-forwarder-utils
(adding lora-picogw-hal-utils compilation return a "world" error, next time i can paste a verbose log)

after that I connected one of my n-Fuse to a USB port, I created a global_conf.json in a generic /etc/lora/ folder, remaining in that path I executed:

root@OpenWrt:/etc/lora# lora_pkt_fwd -d /dev/ttyACM0

On the second attempt it worked regularly, seeing the packets on TTN.
At the first attempt the process remained frozen with:

root@OpenWrt:/etc/lora# lora_pkt_fwd -d /dev/ttyACM0
*** Packet Forwarder for Lora PicoCell Gateway ***
Version: 0
*** Lora concentrator HAL library version info ***
Version: 0;
^CERROR: FAIL TO CONNECT BOARD ON /dev/ttyACM0
INFO: Stopping concentrator

Tomorrow I can try a clean compilation, and give you some more feedbacks. 
Currently I can say that you did a great job, thank you for the support.

closed 2019-07-19 by xueliu