xueliu/lora-feed #8: Compile problem for basicstation

zimiao815, 2019-05-31:
Because I can't download basicstation-2.0.2-52af8aebe0e1edceb68954e78cc494863c747332.tar.gz
 modify Makefile  I get 
basicstation-2.0.3-c29b8502f8c715daecec6666835da6e981dc820a.tar.gz

  **

**> CC    version_features.c
> cc1: note: someone does not honour COPTS correctly, passed 0 times
>   CC    xtea.c
> cc1: note: someone does not honour COPTS correctly, passed 0 times
>   AR    libmbedcrypto.a
> make[8]: NO-TOOLCHAIN-FOUND-ar: Command not found
> make[8]: *** [libmbedcrypto.a] Error 127
> make[7]: *** [lib] Error 2
> make[6]: *** [platform-openwrt/library/libmbedtls.a] Error 2
> make[5]: *** [platform-openwrt] Error 2
> make[5]: Leaving directory `/home/marknum/wangkun/lede/openwrt-17.01.5.arxx/build_dir/target-mips_24kc_musl-1.1.16/basicstation-2.0.3/deps/mbedtls'
> make[4]: *** [deps/mbedtls] Error 2
> make[4]: Leaving directory `/home/marknum/wangkun/lede/openwrt-17.01.5.arxx/build_dir/target-mips_24kc_musl-1.1.16/basicstation-2.0.3'
> make[3]: *** [/home/marknum/wangkun/lede/openwrt-17.01.5.arxx/build_dir/target-mips_24kc_musl-1.1.16/basicstation-2.0.3/.built] Error 2
> make[3]: Leaving directory `/home/marknum/wangkun/lede/openwrt-17.01.5.arxx/feeds/lora/basicstation'
> make[2]: *** [package/feeds/lora/basicstation/compile] Error 2
> make[2]: Leaving directory `/home/marknum/wangkun/lede/openwrt-17.01.5.arxx'
> make[1]: *** [/home/marknum/wangkun/lede/openwrt-17.01.5.arxx/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile] Error 2
> make[1]: Leaving directory `/home/marknum/wangkun/lede/openwrt-17.01.5.arxx'
> make: *** [world] Error 2**

**

xueliu, 2019-06-08:
Hello,

Please follow this tutorial to setup your env
https://clockworkbird9.wordpress.com/2019/03/12/a-simple-test-of-the-brand-new-lora-packet-forwarder-basicstation-in-openwrt/

I will update this package to support CMake.

xueliu, 2019-07-19:
> Because I can't download basicstation-2.0.2-52af8aebe0e1edceb68954e78cc494863c747332.tar.gz
> modify Makefile I get
> basicstation-2.0.3-c29b8502f8c715daecec6666835da6e981dc820a.tar.gz
> 
> **
> 
> **> CC version_features.c
> 
> > cc1: note: someone does not honour COPTS correctly, passed 0 times
> > CC    xtea.c
> > cc1: note: someone does not honour COPTS correctly, passed 0 times
> > AR    libmbedcrypto.a
> > make[8]: NO-TOOLCHAIN-FOUND-ar: Command not found
> > make[8]: *** [libmbedcrypto.a] Error 127
> > make[7]: *** [lib] Error 2
> > make[6]: *** [platform-openwrt/library/libmbedtls.a] Error 2
> > make[5]: *** [platform-openwrt] Error 2
> > make[5]: Leaving directory `/home/lede/openwrt-17.01.5.arxx/build_dir/target-mips_24kc_musl-1.1.16/basicstation-2.0.3/deps/mbedtls' make[4]: *** [deps/mbedtls] Error 2 make[4]: Leaving directory `/home/lede/openwrt-17.01.5.arxx/build_dir/target-mips_24kc_musl-1.1.16/basicstation-2.0.3'
> > make[3]: *** [/home/lede/openwrt-17.01.5.arxx/build_dir/target-mips_24kc_musl-1.1.16/basicstation-2.0.3/.built] Error 2
> > make[3]: Leaving directory `/home/lede/openwrt-17.01.5.arxx/feeds/lora/basicstation' make[2]: *** [package/feeds/lora/basicstation/compile] Error 2 make[2]: Leaving directory `/home/lede/openwrt-17.01.5.arxx'
> > make[1]: *** [/home/lede/openwrt-17.01.5.arxx/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile] Error 2
> > make[1]: Leaving directory `/home/lede/openwrt-17.01.5.arxx'
> > make: *** [world] Error 2**
> 
> **

Hello,

With latest commit 178a3616426e180cab1d2709e74c9f95371550cc I think basicstaion package can be compiled without extra effort.

closed 2019-07-19 by xueliu