xueliu/lora-feed #10: running LoraGW packetforwarder

pacmenDE, 2019-07-30:
Hi,

i´ve tested your feed and compiled succesufully packet_forwarder and libloragw with openwrt 18.06.
for raspi zero.

But i am not able tu start packet forwarder!
Bu i ve done:

/etc/init.d/lora_packet_forwarder
reset pin is on raspi zero is on Gpio 17 (phys. pin11)
So i have changed reset pin to 11, 17 and wiringpi GPIO 0. with reload and restart. but packet_forwarder cannot start.
i´ve chnaged the frequenz from 8000000 to 200000 in menuconfig (libloragw) because the rak2245 concentrator it uses these frequencies. But it´s not running too.

i have an raspbian image with running lora gw und packet forwarder on a second SD Card and it´s running. So it´s not a cabling issue or soo

You have any idea?
Patrik

xueliu, 2019-07-31:
Hello Patrik,

It is great that it finally can be compiled under 18.06.

Please check following things:
1. remove wiringpi. It is not used. It is removed from lora-feed yesterday.
2. check whether `/dev/spi-dev` and `kmod-spi-dev` is there.
3. check `/boot/config.txt` according to https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi

Regards

pacmenDE, 2019-08-01:
Hello Xue,
wiring pi is removed.
spidev exist 
[image]
and at config.txt spi is on.

xueliu, 2019-08-02:
Hello Patrik,
I need further informaion like program log.

pacmenDE, 2019-08-05:
good mornng, 
here the System log.
Mon Aug  5 07:50:41 2019 user.notice root: lora_pkt_fwd: Start
Mon Aug  5 07:50:41 2019 user.notice root: lora_pkt_fwd: Generate /etc/global_conf.json
Mon Aug  5 07:50:41 2019 user.notice root: lora_pkt_fwd: Reset SX1301 with pin 17
Mon Aug  5 07:50:47 2019 user.notice root: lora_pkt_fwd: SX1301 reset completed
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: *** Beacon Packet Forwarder for Lora Gateway ***
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: Version: 0
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: *** Lora concentrator HAL library version info ***
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: Version: 0;
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: ***
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Little endian host
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: found global configuration file /etc/global_conf.json, parsing it
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: /etc/global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: lorawan_public 1, clksrc 1
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: no configuration for LBT
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: antenna_gain 0 dBi
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Configuring TX LUT with 16 indexes
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 129000
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: /etc/global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: gateway MAC address is configured to B827EBFFFE706836
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: server hostname or IP address is configured to "router.eu.thethings.network"
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: upstream port is configured to "1700"
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: downstream port is configured to "1700"
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: downstream keep-alive interval is configured to 10 seconds
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: statistics display interval is configured to 30 seconds
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: upstream PUSH_DATA time-out is configured to 100 ms
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: packets received with a valid CRC will be forwarded
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: packets received with a CRC error will NOT be forwarded
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: INFO: packets received with no CRC will NOT be forwarded
Mon Aug  5 07:50:48 2019 daemon.info lora_pkt_fwd[949]: ERROR: [main] failed to start the concentrator
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: *** Beacon Packet Forwarder for Lora Gateway ***
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: Version: 0
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: *** Lora concentrator HAL library version info ***
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: Version: 0;
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: ***
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: INFO: Little endian host
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: INFO: found global configuration file /etc/global_conf.json, parsing it
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: INFO: /etc/global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
Mon Aug  5 07:50:53 2019 daemon.info lora_pkt_fwd[956]: INFO: lorawan_public 1, clksrc 1
[... cut ...]
Mon Aug  5 07:50:58 2019 daemon.info lora_pkt_fwd[957]: INFO: upstream PUSH_DATA time-out is configured to 100 ms
Mon Aug  5 07:50:58 2019 daemon.info lora_pkt_fwd[957]: INFO: packets received with a valid CRC will be forwarded
Mon Aug  5 07:50:58 2019 daemon.info lora_pkt_fwd[957]: INFO: packets received with a CRC error will NOT be forwarded
Mon Aug  5 07:50:58 2019 daemon.info lora_pkt_fwd[957]: INFO: packets received with no CRC will NOT be forwarded
Mon Aug  5 07:50:58 2019 daemon.info lora_pkt_fwd[957]: ERROR: [main] failed to start the concentrator
Thanks Patrik

pacmenDE, 2019-08-05:
and kernel log:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.184 (patrik@MyPC) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7820-65a4053) ) #0 Tue Jul 23 17:29:54 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 16 MiB at 0x1ac00000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 114688
[    0.000000] free_area_init_node: node 0, pgdat c0541a68, node_mem_map dbc6a700
[    0.000000]   Normal zone: 896 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113792
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:25:3D:63 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 432480K/458752K available (3841K kernel code, 152K rwdata, 1076K rodata, 156K init, 348K bss, 9888K reserved, 16384K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc03c8988   (3843 kB)
[    0.000000]       .init : 0xc04f5000 - 0xc051c000   ( 156 kB)
[    0.000000]       .data : 0xc051c000 - 0xc0542240   ( 153 kB)
[    0.000000]        .bss : 0xc0542240 - 0xc05992ac   ( 349 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000028] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000073] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000160] bcm2835: system timer (irq = 27)
[    0.000643] Console: colour dummy device 80x30
[    0.001380] console [tty1] enabled
[    0.001446] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060316] pid_max: default: 32768 minimum: 301
[    0.060501] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060562] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061342] CPU: Testing write buffer coherency: ok
[    0.061750] Setting up static identity map for 0x8200 - 0x825c
[    0.070843] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.071330] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.071432] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.073618] pinctrl core: initialized pinctrl subsystem
[    0.074997] NET: Registered protocol family 16
[    0.077378] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.085711] cpuidle: using governor ladder
[    0.085801] cpuidle: using governor menu
[    0.086307] Serial: AMBA PL011 UART driver
[    0.090731] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.091438] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.131598] bcm2835-dma 20007000.dma: DMA legacy API manager at dc80d000, dmachans=0x1
[    0.134722] SCSI subsystem initialized
[    0.135028] usbcore: registered new interface driver usbfs
[    0.135179] usbcore: registered new interface driver hub
[    0.135322] usbcore: registered new device driver usb
[    0.140699] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-11-23 19:36
[    0.142265] clocksource: Switched to clocksource timer
[    0.152880] NET: Registered protocol family 2
[    0.153851] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[    0.153987] TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
[    0.154098] TCP: Hash tables configured (established 4096 bind 4096)
[    0.154220] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.154288] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.154567] NET: Registered protocol family 1
[    0.156032] Crashlog allocated RAM at address 0x3f00000
[... 124 lines cut ...]
[    9.730692] nf_conntrack version 0.5.0 (7168 buckets, 28672 max)
[    9.793301] input: CHICONY USB Keyboard as /devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.0/0003:04F2:0833.0001/input/input0
[    9.882828] hid-generic 0003:04F2:0833.0001: input,hidraw0: USB HID v1.11 Keyboard [CHICONY USB Keyboard] on usb-20980000.usb-1/input0
[    9.919510] input: CHICONY USB Keyboard as /devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.1/0003:04F2:0833.0002/input/input1
[... cut ...]
[   10.590521] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-11), device may have limited channels available
[   10.615889] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug 29 2016 20:48:16 version 7.45.41.26 (r640327) FWID 01-4527cfab
[   10.678120] kmodloader: done loading kernel modules from /etc/modules.d/*
[   13.869827] random: crng init done
[   13.881780] random: 3 urandom warning(s) missed due to ratelimiting
[   21.934683] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   23.107523] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

xueliu, 2019-08-08:
It is hard to say where is the problem. You may use the utils together with libloragw to find the problems.

pacmenDE, 2019-08-21:
Hi Xue,
now it runs! :-)

But with some modification... :-)

Over Luci > Network > Lora...
I can not start lora packet forwarder... Here the screenshot
[image]

This helped me to run lora packet forwarder.
1.) manual configured the /etc/global_config.json
2.) /usr/sbin/ create a lora_reset.sh fo reseting SX1301 (based on this script https://raw.githubusercontent.com/Lora-net/lora_gateway/master/reset_lgw.sh)
3. modified the sleep function from 0.1 to 1 
4. chmod 0777 reset.sh
5  run reset sript
6 start lora_packet_forwarder

and it runs.... :-)

pacmenDE, 2019-08-25:
changed "/etc/init.d/lora_packet_forwarder" solved the problem!
look for pull request

closed 2019-08-25 by pacmenDE