xueliu/lora-feed #12: commit  52d39b0 error qsort_r

pacmenDE, 2019-08-06:
hi xue,
i´ve updated the feeds yesterday. now i´get compile error from lora_packet forwarder. The problem ist that the semteck source code for packet forwarder has "qsort_r" function and it look like that this function is not able to compile with openwrt > Version 15.05.
Here the compile message:
arm-openwrt-linux-muslgnueabi-gcc -L../../lora_gateway-5.0.1/libloragw obj/lora_pkt_fwd.o obj/parson.o obj/base64.o obj/jitqueue.o obj/timersync.o -o lora_pkt_fwd -lloragw -lrt -lpthread -lm
obj/jitqueue.o: In function `jit_sort_queue':
jitqueue.c:(.text+0x134): undefined reference to `qsort_r'
collect2: error: ld returned 1 exit status
Makefile:65: recipe for target 'lora_pkt_fwd' failed
make[5]: *** [lora_pkt_fwd] Error 1
make[5]: Leaving directory '/home/patrik/openwrt1806/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/packet_forwarder-4.0.1/lora_pkt_fwd'
Makefile:11: recipe for target 'all' failed
make[4]: *** [all] Error 2

xueliu, 2019-08-08:
Hello Patrik,

The patch for the the qsort_r is already there.
https://github.com/xueliu/lora-feed/blob/master/lora-packet-forwarder/patches/002-using_qsort.patch

pacmenDE, 2019-08-21:
Hi Xue,
update the feeds ficed the problem.
Thanks Patrik

closed 2019-08-21 by pacmenDE