xueliu/lora-feed #19: Basicstation 2.0.5 and USB GW devices

donetdisturb, 2020-06-14:
Hello Xue Liu,
https://github.com/lorabasics/basicstation/releases Basicstation 2.0.5 (with support for ACM USB devices) is out, it's possible to align the code for compiling? Thank you for you work

xueliu, 2020-06-23:
Hello,

I am on the way of porting 2.0.5 to OpenWRT. The newest version has three different HALs, which needs a lot of time to adapt.

Do you mind to tell me which LoRaWAN gateway hardware do you use ? e.g. sx1301 + RPi.

donetdisturb, 2020-06-23:
Hello Xue, I'm compiling OpenWRT in x86 image for Geode CPU (ALIX board)

s-2, 2020-06-23:
This looks awesome, it seems I could finally use my picoGW card with GPS and Class B devices.
Have you tried compiling already, and it failed? I wonder how much adaption is required for this to work on OpenWRT.

xueliu, 2020-06-24:
Hello all,

I have tried to bump the basicstation version to 2.0.5.
I added a option in the package to select one from threeHALs.
But you have to input it by hand. e.g. linuxpico for SX1308 pico gateway.
The work I have done is only binary compiling, other scripts e.g. /etc/init.d/ and UCI are not adapted jet.
Therefore you should follow the instruction from https://doc.sm.tc/station/ to setup basicstation.

s-2, 2020-06-24:
Thank you!

I can't seem to successfully compile mbedtls though, not sure whether it's related to basicstation at all. I also unsinstalled lora-gateway-hal from the official feeds and replaced with yours, so that could probably not cause the error?

The error comes from compiling bignum.c.o:

```
...
-- Build files have been written to: /home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download
make[4]: Entering directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download'
make[5]: Entering directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download'
make[6]: Entering directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download'
make[6]: Leaving directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download'
make[6]: Entering directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download'
[ 11%] Performing build step for 'mbedtls'
make[7]: Entering directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-build'
make[8]: Entering directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-build'
make[9]: Entering directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-build'
make[9]: Leaving directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-build'
make[9]: Entering directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-build'
[  0%] Building C object library/CMakeFiles/mbedcrypto.dir/bignum.c.o
{standard input}: Assembler messages:
{standard input}:130: Error: invalid operands `lw $10,12($sp)'
{standard input}:131: Error: invalid operands `lw $11,16($sp)'
{standard input}:132: Error: invalid operands `lw $12,4($sp)'
{standard input}:133: Error: invalid operands `lw $13,20($sp)'
{standard input}:134: Error: invalid operands `lw $14,0($10)'
{standard input}:135: Error: invalid operands `multu $13,$14'
{standard input}:136: Error: unrecognized opcode `addi $10,$10,4'
```
.... and a lot more of these errors ...
```
 library/CMakeFiles/mbedcrypto.dir/build.make:157: recipe for target 'library/CMakeFiles/mbedcrypto.dir/bignum.c.o' failed
make[9]: *** [library/CMakeFiles/mbedcrypto.dir/bignum.c.o] Error 1
make[9]: Leaving directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-build'
CMakeFiles/Makefile2:680: recipe for target 'library/CMakeFiles/mbedcrypto.dir/all' failed
make[8]: *** [library/CMakeFiles/mbedcrypto.dir/all] Error 2
make[8]: Leaving directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-build'
Makefile:157: recipe for target 'all' failed
make[7]: *** [all] Error 2
make[7]: Leaving directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-build'
CMakeFiles/mbedtls.dir/build.make:130: recipe for target 'mbedtls-prefix/src/mbedtls-stamp/mbedtls-build' failed
make[6]: *** [mbedtls-prefix/src/mbedtls-stamp/mbedtls-build] Error 2
make[6]: Leaving directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download'
CMakeFiles/Makefile2:92: recipe for target 'CMakeFiles/mbedtls.dir/all' failed
make[5]: *** [CMakeFiles/mbedtls.dir/all] Error 2
make[5]: Leaving directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download'
Makefile:100: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/mbedtls/mbedtls-download'
CMake Error at deps/mbedtls/CMakeLists.txt:16 (message):
  Build step for mbedtls failed: 2

-- Configuring incomplete, errors occurred!
See also "/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/CMakeFiles/CMakeOutput.log".
Makefile:70: recipe for target '/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/sebastian/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/sebastian/openwrt/openwrt/feeds/lora/basicstation'
time: package/feeds/lora/basicstation/compile#0.69#0.08#0.74
package/Makefile:111: recipe for target 'package/feeds/lora/basicstation/compile' failed
make[2]: *** [package/feeds/lora/basicstation/compile] Error 2
make[2]: Leaving directory '/home/sebastian/openwrt/openwrt'
package/Makefile:107: recipe for target '/home/sebastian/openwrt/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/sebastian/openwrt/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sebastian/openwrt/openwrt'
/home/sebastian/openwrt/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
```

Could you check compiling for the MT7620 target with the `linuxpico` platform?
Maybe you could figure out what's wrong, maybe I'm missing some dependencies, or could this be due to the endianness of the platform?

xueliu, 2020-06-25:
> Thank you!
>
> I can't seem to successfully compile mbedtls though, not sure whether it's related to basicstation at all. I also unsinstalled lora-gateway-hal from the official feeds and replaced with [...]

The problem comes from compiling of local libmbedtls with OpenWRT MIPS toolchain . I have update the lora-feed with a quick fix using libmbedtls from OpenWRT. I have tested the code with MT7260 platfrom.

s-2, 2020-06-25:
Thanks, it compiled perfectly now, I will test later on the weekend :)

donetdisturb, 2020-06-27:
Hello Xue, I can confirm, it works. :-)

Setup:
OpenWRT (latest stable tag) for x86 GEODE, compiling on Ubuntu 18.04 LTS.
Pkg: basicstation "linuxpico"
Device: RADIODEV=/dev/ttyACM0
Hardware: n-fuse sx1301

Finally the Basicstation on WRT! Thank you!

xueliu, 2020-07-02:
I think I can close this issue.

closed 2020-07-02 by xueliu

s-2, 2021-03-07:
Sorry for digging this out again, I've only recently found the time to re-setup everything regarding my gateway, trying to take it to TheThings Stack V3 (for Class B), but there seem to be too many disadvantages yet (no ttnmapper etc.), so I'm still with V2.

Just a few things I had noticed, which are probably mostly due to my lack of understanding of the available configuration settings:
- On TTN coverage map, the Gateway "model" string is fixed to `linuxpico - Firmware - Protocol 2`
  could someone point me to the where this setting can be changed, e.g. in the `station.conf` or somewhere before compilation?
- The Gateway EUI used is determined by station from the MAC address of the first ethernet interface it can find in OpenWRT. Unfortunately, OpenWRT will often use `base-files` scripts that change uci default settings for interfaces (e.g. eth0.1) to override MAC address with the correct one for the device, while the eth0 device itself would stick with the randomized address chosen on boot, when no valid MAC address could be determined.
  Unfortunately, this means that my Gateway would report a randomized EUI to the Network Server on every boot, so it would lose connectivity every time (besides the invalid mac used); could someone point me to the setting where I can override the Gateway EUI? I think it should be in `global.conf` or `station.conf`, but I've seemed to miss it from the little and confusing documentation provided by Semtech.
- Since I've successfully configured the PPS signal input, the output of station is spammed with messages "Timesync #nn sent to Server". Could someone elaborate on that, i.e. is it actually sending several of these packets per second to TTN servers? Is this intended or erroneous behavior / what is it good for? (occurs way more often than actual uplink / join frames received) 
- Every 24h after an IP address change from our ISP, the reporting to TTN is broken and I have to restart station. However, it is still running and reporting received frames to the shell output, but just no longer forwarding to TTN.
  Maybe this is an issue of client authentication, I think I might have to generate an actual client certificate here, in addition to placing the server public certificate in the /etc/station folder.

@donetdisturb Have you maybe encountered any of these issues as well, what is your general experience with basicstation so far?

I'm not sure whether this is the right place to ask, I'm just very confused by the lack of practical information out there, when it comes to building your own gateway with the latest technology (rather than pkt_forwarder)...