xueliu/lora-feed #20: Is there any plans to update the lorawan-stack package recently?

guo652917087, 2020-07-28:
Hi,
    I tried to update the version of lorawan-stack V3.9, and it seems that the front-end page has not been compiled into ttn-lw-stack. I used go V1.14, modified your compilation steps:

@echo "make init"
@make init -C $(PKG_BUILD_DIR)
(\
export GOPATH=$(GO_PKG_BUILD_DIR) \
GOCACHE=$(GO_PKG_CACHE_DIR) \
GOTMPDIR=$(GO_PKG_TMP_DIR) \
GOROOT_FINAL=$(GO_TARGET_ROOT) \
CC=$(TARGET_CC) \
CXX=$(TARGET_CXX) \
$(call GoPackage/Environment) \
\
&& cd $(PKG_BUILD_DIR) \
&& tools/bin/mage js:clean js:build \
&& echo "Compiling CLI..." \
&& go build ./cmd/ttn-lw-cli \
&& echo "Compiling Stack..." \
&& go build ./cmd/ttn-lw-stack \
)

The test log shows that the page does not exist（http:404）:

duration=199.011 method=GET namespace=web remote_addr=192.168.2.25 request_id=01EE9PX51BYTYYWD2MR4S4PRPH response_size=19 status=404 url=http://192.168.2.1:1885/assets/console.js

Do you have any way to compile front-end js?

xueliu, 2020-08-07:
Hello,

Yes. the front-end is not compiler at all. Because it uses local program called mage to compile, which is not sucessfully used in the Makefile.

I use the blog https://blog.csdn.net/iotisan/article/details/91954463 to make the Makefile.

guo652917087, 2020-08-08:
Hi,
    Is it possible to use goreleaser?

xueliu, 2020-08-08:
Hello,

Please have a try with the latest release ae345b0fde87cc20d52c0876b41005bdd4c58d41.

The binary seems working. I did not success to cross compile the cockroach. so I can not continue to test.

guo652917087, 2020-08-10:
Hi,
  No work.I use postgresql data test, 192.168.2.148 is the openwrt host, 192.168.2.150 is my ubuntu host, and the ubuntu log is also below.
[image]

[image]

[image]

closed 2020-08-10 by guo652917087