xueliu/lora-feed #21: pico_pkt_fwd not starting at boot as it should

artynet, 2020-10-23:
I'm currently using the plain **lora-packet-forwarder** init file based on procd tailored to the **pico_pkt_fwd** :

the main change is basically the change of the executable name from l**ora_pkt_fwd** to **pico_pkt_fwd**

```
#!/bin/sh /etc/rc.common
START=99
STOP=10

USE_PROCD=1

PROG=/usr/sbin/pico_pkt_fwd
CONFIGFILE=/etc/lora/global_conf.json
```
anyway, after adding a function to check the online status :

```
check_online_status()
{
	intertube=0
	echo "Verifying ping(s) for interface wwan0..."
	while [ $intertube -ne 1 ]; do
        ping -c5 google.com
        if [ $? -eq 0 ];then
                echo "Correctly connected to the 4g modem"
        echo ${pin} > /sys/class/gpio/export
        echo "out" > /sys/class/gpio/gpio${pin}/direction
        sleep 1
	fi
}
```
the process is stuck to and the concentrator doesn't start :
```
Fri Oct 23 16:45:44 2020 daemon.info pico_pkt_fwd[3202]: *** Packet Forwarder for Lora PicoCell Gateway ***
Fri Oct 23 16:45:44 2020 daemon.info pico_pkt_fwd[3202]: Version: 0
Fri Oct 23 16:45:44 2020 daemon.info pico_pkt_fwd[3202]: *** Lora concentrator HAL library version info ***
Fri Oct 23 16:45:44 2020 daemon.info pico_pkt_fwd[3202]: Version: 0;
```
any ideas about it ?

xueliu, 2020-10-23:
Hello Arturo,

Sorry I have no idea about this problem. Please attach the full init script and the OpenWRT version. Thanks.

artynet, 2020-10-23:
Hello @xueliu, here's the code :

```
#!/bin/sh /etc/rc.common
START=99
STOP=10

USE_PROCD=1

PROG=/usr/sbin/pico_pkt_fwd
CONFIGFILE=/etc/lora/global_conf.json

reset_sx1301_board ()
{
        local pin=$1

        logger "pico_pkt_fwd: Reset SX1301 with pin ${pin}"

        if [ -d "/sys/class/gpio/gpio${pin}" ]
        then
                echo 1 > /dev/null
        else
                echo ${pin} > /sys/class/gpio/export
                echo "out" > /sys/class/gpio/gpio${pin}/direction
                sleep 1
        fi

        echo "1" > /sys/class/gpio/gpio${pin}/value
        sleep 5
        echo "0" > /sys/class/gpio/gpio${pin}/value
        sleep 1
        echo "0" > /sys/class/gpio/gpio${pin}/value
        sleep 1
        echo "in" > /sys/class/gpio/gpio${pin}/direction
        sleep 1

        logger "pico_pkt_fwd: SX1301 reset completed"
}

generate_global_conf()
{
        logger "pico_pkt_fwd: Generate ${CONFIGFILE}"
        gen_lora_global_conf > ${CONFIGFILE}
}

check_lora_device()
{
        while [ ! -c /dev/ttyLORA ];do
		echo 'waiting for interface to be raised....'
		sleep 1
        done
}

check_online_status()
{
        intertube=0
        echo "Verifying ping(s) for interface wwan0..."
        while [ $intertube -ne 1 ]; do
		ping -c5 google.com
		if [ $? -eq 0 ];then
			echo "Correctly connected to the 4g modem"
			intertube=1
		fi
        done
}

start_service()
{
        include /lib/functions

        logger "pico_pkt_fwd: Start"

        # config_load pico_pkt_fwd

        check_lora_device

        check_online_status

        # generate_global_conf

        local reset_enable=$(uci get lora-global.SX1301_conf.enable_reset_pin)

        # if [ ${reset_enable} == 1 ]; then
        #       reset_sx1301_board $(uci get lora-global.SX1301_conf.reset_pin)
        # fi

        procd_open_instance picogw
        procd_set_param command $PROG
        procd_set_param file ${CONFIGFILE}
        procd_set_param file /etc/config/lora-global
        procd_set_param pidfile /var/run/lora.pid
        procd_set_param respawn

        procd_set_param stdout 1
        procd_set_param stderr 1

        procd_close_instance
}

stop_service() {
    logger "pico_pkt_fwd: Stop"
    killall pico_pkt_fwd
}
```
I have patched the source code of the picoGW binary so to have as default device **/dev/ttyLORA** instead of **/dev/ttyACM0** which is  an UDEV symlink I did on purpose to avoid the device character renaming (i.e. **/dev/ttyACM1**, **/dev/ttyACM2** and so on) in case of software reset....

thanks in advance...

xueliu, 2020-11-02:
Hello,

I did not see any problem in your init script.
Can you start pico_pkt_fwd from shell ? Can it running normally ?

artynet, 2020-11-05:
hello @xueliu it seems it was a matter of updating the mcu firmware :

https://www.hackster.io/fewi/build-a-lora-gateway-with-n-fuse-mpcie-card-71f0e1

as explained here :

https://github.com/Lora-net/picoGW_mcu#2-precompiled-binaries

after doing that the system works smoothly !

closed 2020-11-12 by xueliu