xueliu/lora-feed #22: loragwhal : spidev change and library debug

queifaro, 2020-12-10:
Hello,
how to change spidev ? seems that change in make menuconfig do not affect the real spidev used (remail spidev0.0).
Also how to obtail a compilation with library debug mode ON ? changing library.cfg seems do not affect the compilation and config.h remain the same.
Thanks in advance
B/R
Fabio

xueliu, 2020-12-17:
Hello Fabio,

About the spidev, I will check it again.
About the library debug mode, it is currently only supported by out of tree build and not supported in OpenWRT menuconfig. I will push a new version to add debug options.

Regards

Xue Liu

queifaro, 2020-12-21:
Hello Xue,
thanks for reply. Please forget the spidev issue , was my stupid problem.
Library debug mode it will be helpful.
Thanks
B/R
Fabio

techatreyo, 2024-06-05:
> Hello Xue, thanks for reply. Please forget the spidev issue , was my stupid problem. Library debug mode it will be helpful. Thanks B/R Fabio

Are you able to use this luci app for connecting gateway to LNS server ?????