xueliu/lora-feed #23: Support SX1302/SX1303 Concentrators

chrissnow, 2021-02-12:
Is it possible to support the newer concentrator cards?
Semtech seem to have a different repo for everything.
https://github.com/Lora-net/sx1302_hal

chrissnow, 2021-02-19:
@xueliu 

I have made some progress and basicstation seems to be to the way to go but I'm really struggling with patching qsort
If I define CONFIG_GATEWAY_PLATFORM:=corecell
I end up with
```
make[8]: Entering directory '/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-build'
[ 86%] Building C object packet_forwarder/CMakeFiles/lora_pkt_fwd.dir/src/jitqueue.c.o
/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-src/packet_forwarder/src/jitqueue.c: In function 'jit_sort_queue':
/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-src/packet_forwarder/src/jitqueue.c:117:5: warning: implicit declaration of function 'qsort_r'; did you mean 'qsort'? [-Wimplicit-function-declaration]
     qsort_r(queue->nodes, queue->num_pkt, sizeof(queue->nodes[0]), compare, &counter);
     ^~~~~~~
     qsort
[ 88%] Building C object packet_forwarder/CMakeFiles/lora_pkt_fwd.dir/src/lora_pkt_fwd.c.o
[ 91%] Linking C executable lora_pkt_fwd
/onion/openwrt19_6/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/7.5.0/../../../../mipsel-openwrt-linux-musl/bin/ld: CMakeFiles/lora_pkt_fwd.dir/src/jitqueue.c.o: in function `jit_sort_queue':
jitqueue.c:(.text+0xb4): undefined reference to `qsort_r'
collect2: error: ld returned 1 exit status
make[8]: *** [packet_forwarder/CMakeFiles/lora_pkt_fwd.dir/build.make:100: packet_forwarder/lora_pkt_fwd] Error 1
make[8]: Leaving directory '/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-build'
make[7]: *** [CMakeFiles/Makefile2:546: packet_forwarder/CMakeFiles/lora_pkt_fwd.dir/all] Error 2
make[7]: Leaving directory '/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-build'
make[6]: *** [Makefile:130: all] Error 2
make[6]: Leaving directory '/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-build'
make[5]: *** [CMakeFiles/lora_gateway.dir/build.make:114: lora_gateway-prefix/src/lora_gateway-stamp/lora_gateway-build] Error 2
make[5]: Leaving directory '/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-download'
make[4]: *** [CMakeFiles/Makefile2:76: CMakeFiles/lora_gateway.dir/all] Error 2
make[4]: Leaving directory '/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-download'
make[3]: *** [Makefile:84: all] Error 2
make[3]: Leaving directory '/onion/openwrt19_6/build_dir/target-mipsel_24kc_musl/basicstation-2.0.5/deps/lgw1302/sx1302_hal-download'
CMake Error at deps/lgw1302/CMakeLists.txt:16 (message):
  Build step for sx1302_hal failed: 2

```

I have the following patch which will fix it, but I'm going around circles on how to implement it.

```
diff --git a/packet_forwarder/src/jitqueue.c b/packet_forwarder/src/jitqueue.c
index 22ec701..748843d 100644
--- a/packet_forwarder/src/jitqueue.c
+++ b/packet_forwarder/src/jitqueue.c
@@ -112,8 +112,8 @@ void jit_sort_queue(struct jit_queue_s *queue) {
     }

     MSG_DEBUG(DEBUG_JIT, "sorting queue in ascending order packet timestamp - queue size:%u\n", queue->num_pkt);
-    qsort_r(queue->nodes, queue->num_pkt, sizeof(queue->nodes[0]), compare, &counter);
-    MSG_DEBUG(DEBUG_JIT, "sorting queue done - swapped:%d\n", counter);
+    qsort(queue->nodes, queue->num_pkt, sizeof(queue->nodes[0]), compare);
+    MSG_DEBUG(DEBUG_JIT, "sorting queue done\n");
 }

 bool jit_collision_test(uint32_t p1_count_us, uint32_t p1_pre_delay, uint32_t p1_post_delay, uint32_t p2_count_us, uint32_t p2_pre_delay, uint32_t p2_post_delay) {

```

Any help would be greatly appreciated.

xueliu, 2021-02-20:
Hello Chris,

I will check this problem in these days.

xueliu, 2021-02-20:
Hello Chris,

After a quick check, I found that now it is not necessary to update sx1302 to the latest version because the basicstation did not update jet. It still use the v1.0.5 and the patch https://github.com/xueliu/basicstation/blob/feature/add-cmake/deps/lgw1302/V1.0.5-corecell.patch can not apply to the latest sx1302 repo.

I would like to know which hardware platform do you use and how to get a chep sx1302/sx1303 concentrator.

Regards,

Xue Liu

chrissnow, 2021-02-20:
Hi Xue,

Thanks for looking into it, all help is appreciated!

I'm trying to get it working with an Onion Omega 2+ dev board, I'm not aware of a cheap source of them yet though, I had to buy the card from Digikey, I think a Picocell might be sensible money though.

I also tried cross compiling basicstation directly from Semtech, which compiles but after 1 line of console output it crashes, without a kernel panic or anything and needs a physical reset, the serial console goes dead.

If you can figure out how to get it to build I will gladly test it.

I'm new to building OpenWrt and struggling with the build tools a bit, especially with how the repos seem to checkout repos and then switch to branches.

xueliu, 2021-02-21:
Hello Chris,

I think the latest commit can solve the problem.

I have reproduced the same problem with the platform setting corecell. The reason is that the 'qsort_r' is not support in the musl library. It is already fixed in the program packet_forwarder https://github.com/xueliu/lora-feed/blob/master/lora-packet-forwarder/patches/002-using_qsort.patch. 

I fix the same problem for basicstation by adding a new branch 'openwrt' in my sx1302-hal fork https://github.com/xueliu/sx1302_hal/commit/17173223c2284c7379c49a3d3ef580b1fea96b30

The "repos checkout repos" has nothing to do with OpenWRT. The CMake did it actually.

Please let know if it is working.

chrissnow, 2021-02-22:
I have been able to build and tried to test it, but it's not that simple it seems :-(
https://github.com/lorabasics/basicstation/issues/89#issuecomment-783226399

The version of libloragw that basicstation supports is too old for the USB version.

I will need to try and build this again it seems.
https://github.com/Lora-net/sx1302_hal

What's the best way to add this to the OpenWrt build system?
I will try debugging with GDB why my directly cross compiled version crashes.

This is how I am building it at the moment.
```
cd /onion/sx1302_hal
export CROSS_COMPILE=mipsel-openwrt-linux-musl-
export ARCH=mips
PATH=$PATH:/onion/openwrt19_6/staging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/bin/
export PATH
export STAGING_DIR=/onion/openwrt19_6/staging_dir/
make clean
make
```

xueliu, 2021-02-22:
Hi,

It is out of the scope of feed-lora. If you really want to build basicstaiton out of the OpenWRT tree with the original makefile. I can share my experience two years ago, which may not apply 2.0.5. https://clockworkbird9.wordpress.com/2019/03/12/a-simple-test-of-the-brand-new-lora-packet-forwarder-basicstation-in-openwrt/

closed 2021-02-22 by xueliu

chrissnow, 2021-02-22:
A bit of an update, cross compiling with my above commands does work, and results in a working gateway.

I meant how to add sx1302_hal to lora-feed rather than compile basicstation out of tree.

xueliu, 2021-02-22:
I will add it later.