xueliu/lora-feed #33: basicstation compile variant

queifaro, 2022-07-20:
Hello,
Semtech basicstation can be configured using some kind of platform and for each platform (linux,corecell,linuxpico...) some kind of variants (**std,stdn, ...**).
For example variant stdn enable the ral maste/slave (two lora hw board).

In the last commit, openWrt makefile seems provide multiple package of basicstation depending on platform (linux, corecell ecc) . Also seeams that this depend on the variable VARIANT that makefile provide.

Is VARIANT really and only mean platform ? howto compile chosing, for example,  Semtech variant **stdn** ?

Thanks in advance
B/R
Fabio

s-2, 2023-08-20:
Hi everyone,

in attemtping to upgrade my gateway to a more current version, I tried building `basicstation-linuxpico`, but could not select this in `menuconfig` anywhere, there is just a generic entry for `basicstation` without the variants.
Patching the image Makefile of the device I use to include that package makes it show up in the config file, but still the regular basicstation (trying to connect via spi) would be contained within the image. Calling `make clean` and not selecting basicstation in menuconfig results in no `station` binary in the image at all, despite being included via `DEVICE_PACKAGES`.

Any advice on how to build the `linuxpico` variant after the changes introduced a while ago?
Back then, I recall there was a little sub-menu for selecting the variant, but this is no longer visible.
So, what would be the intended way to actually build variants other than the default `-linux`?

Thanks,
Sebastian

s-2, 2023-08-20:
Okay I remembered now,
I forgot one thing I also had to do when initially setting this up back a couple years ago,
and that's removing the `basicstation` package from OpenWrt's feed first.

`./scripts/feeds uninstall basicstation`

and afterwards, as in the Readme:

`./scripts/feeds install -a -p lora`

will add the correct basicstation (and the variants for selection in menuconfig) from the lora feed, rather than the OpenWrt feed. However I couldn't see `std`, `stdn` variants, what are those meant for?

queifaro, 2023-08-20:
Std and Stdn is for one/ multiple boardsWhy do you want to uninstall OpenWrt basicstation ?Fabio

s-2, 2023-08-20:
> Why do you want to uninstall OpenWrt basicstation ?

I could not select the `basicstation-linuxpico` variant in menuconfig otherwise, when basicstation from OpenWrt feed was still present (trying to connect via SPI, though I use the picocell design with the USB MCU).

I actually wanted to update to 2.0.6 to fox Class B Beacon hadling, but it seems the feed still points to 2.0.5.
Can I just change it from 9bcdf0e12e4568e96011a3910e6d460ec9384f8c to latest master to get the current version, or is more adaptation required here? (this should probably be discussed in a new issue though)

queifaro, 2023-08-20:
I suggest to use OpenWrt basicstation that already point to 2.0.6 and adapt it to pico gw compile variant configuration.